sorted01/Filametrix #4: Rapido CAD/STL

juliusjj25, 2023-06-20:
Not an issue, but I took your CAD modifications and applied them to the Rapido Toolhead design and created the CW2 main body for the ERCF w/ ECAs, bearing switch, and SB2209 Canbus mods

@CheeseFrog V2.4559

[Rapido & SB2209.zip](https://github.com/sorted01/Filametrix/files/11795457/Rapido.SB2209.zip)

sorted01, 2023-06-20:
Thank you so much! :) I've added it to the files and updated the readme. I hope this is fine!

I would close the "not an issue" :D

closed 2023-06-20 by sorted01

akillmer, 2023-06-23:
@juliusjj25 Which PCB cover are you using for this? The original cover leaves a considerable gap.

juliusjj25, 2023-06-24:
@akillmer I am using the one from the BTT SB2209 design on Github
 https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/Custom%20Printed%20Parts/cable_door_EBB_SB.stl

jgorges7, 2023-08-14:
@juliusjj25 Does this use a 5.5mm ball bearing for the sensor? Better yet, do you have a step file for the main body? I do not intend to use the filament sensor and would like to remove the added resistance from the system. Thanks!

juliusjj25, 2023-08-14:
> @juliusjj25 Does this use a 5.5mm ball bearing for the filament sensor? Better yet, do you have a step file for the main body? I do not intend to use the filament sensor and would like to remove the added resistance from the system. Thanks!

This uses the 7mm bearing ball from the ERCF usermods. I appear to have misplaced the step file for these mods.