sorted01/Filametrix #7: The dual D2F SB part has a major flaw

scottgrim, 2023-07-16:
When the part for the heat insert was added at the exit path of the CW2 the entire filament path between the last sensor and the insert was not reduced to 2mm so there's a lip inside the part for filament that has any amount of curl to catch on.

[image]

That circled section should be 2mm straight through.

scottgrim, 2023-07-16:
Something like this which I just hacked together in TC real quick.
[image]

sorted01, 2023-10-09:
Sorry, missed that one. You're right. As i did not design this part and i have no step i'll delete it for now. Could you share your "quickfix"?

sorted01, 2023-11-06:
It's solved :)