sorted01/Filametrix #8: Crazy Dragon (Dragon UHF mini mode)

messymouse, 2023-07-18:
I would love a version for Crazy Dragon, which is a Dragon UHF without the nozzle extender part on it. Crazy Dragon mount can be found here- https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/eamars/stealthburner_crazy_dragon_toolhead/STL

Its basically a combination of Rapido and Dragon, which are both already created here, I just have no idea how to do the 3D modeling to merge them correctly. Thanks!

ntsanov, 2023-07-24:
I'm using crazy dragon, basically a Phaetus Dragon SF with a Mellow Crazy Volcano block. I'm also interested. Since the step files for the Crazy Dragon are released it should be straight forward. I might do it myself when I get to the ERCF at some point in the following months

messymouse, 2023-07-27:
I don't know anything about 3D modeling, and every time I have tried figuring out how to use a step file it was a headache and I got no where. This seems like a simple thing to throw together for people who know what they are doing, and I would greatly appreciate it!

EmJay276, 2023-12-14:
I just added versions for the Phaetus Dragon UHF and Phaetus Dragon UHF mini :)
https://github.com/sorted01/Filametrix/pull/36