sorted01/Filametrix #11: Solution for the gantry mounted part for people with extrusion backers?

erazor666, 2023-08-02:
any ideas on how to attach the gantry part of the cutter mod when you have  extrusion backers on your Y axis?
I really want to keep them and use this mod :(

o1hitman1o, 2023-08-03:
I would say is it possible to put the backers on the exterior side instead of the top? I don't have the backers installed myself but if it will fit I would think on the side would offer the same performance at on the top.

erazor666, 2023-08-07:
I would think the backers will not have the optimum performance of counteracting the bi-metallic effect that they are there for :) I guess that it is quite possible to design something in Fusion360 when I get a free minute. Was just looking for the possibility that someone else had this same issue :)

erazor666, 2023-08-15:
I guess people have different backers, I myself am using the rails of my old MGN9's that used to be on my V2 X axis. since they were rather useless after converting to MGN12 on the X rail. There are some premade backers that are titanium and such, dont know if they have same screw hole pattern as a MGN rail. What are you basing the design on? From a purely visual standoint this seems very much the way to go. (and the way I was doing it on my napkin design ) :)

juliusjj25, 2023-08-16:
It's my own design from scratch, I have titanium backers that I am
designing around but I do not have them installed yet, which is why I was
wondering about the positioning. The titanium backers have 40mm hole
spacing and they have varied distances from the edge of the backer 5-20mm
based on the build size. I do not know the hole spacing on the MGN9 rails
and that may or may not be standard across rails. If it helps, when I
finish the design I could share the step or native Solidworks file for you
to modify to your needs, likely you would just need to change the
overall height and the hole spacing and relationship to the edge.

juliusjj25, 2023-08-16:
I took some measurements and figured it out and adjusted the CAD. I had to make it wider because the backer hole aligned under the 2020 piece. I also made a 250, 300 & 350 build version. Here is the 300 version. 
[image]

SLDPRT, STEP & STL files attached. :) Print/modify at your own risk, I have not done any sort of testing or printing of any of this yet. I am waiting on parts to build this up along with several other mods, so I won't get around to testing for several weeks. 
[2020_mounts_ti_backer.zip](https://github.com/sorted01/Filametrix/files/12354111/2020_mounts_ti_backer.zip)

erazor666, 2023-08-16:
I will have a look at this in the afternoon, it looks like the way to go very much. Hole spacing on a MGN9 is 20mm it seems from measuring. Probably this would be worthwhile to do a pull request on eventually, there are many people who use backers (and should)

juliusjj25, 2023-08-16:
If your hole spacing is 20mm then this should work for you as well. 

I printed one of these last night since my printer wasn't running, I have some tweaks to make before I'd say this is ready to go. The t-slot profile needs more clearance and I want to tighten up the holes at the top because the screws don't quite bite enough to my liking, maybe add a gusset or to to stiffen it up. 

I might design a full 3D printed part to make this a little cleaner, I am not sure how much force it takes, but given the lever arm thickness, it can't be that much.

I'll see where I get on this today, I think it will still need a build size variant of each unless I make it slotted with some notches to align the screws to.

juliusjj25, 2023-08-16:
This is what I have come up with this morning as a 2 piece design. The mount spacing is 40mm and it has 10mm slots, so it can be adjusted front to back 5mm from nominal, which is the difference between the 250, 300 & 350 build sizes. For titanium backers, it should be aligned with the back of the backer. The BHCS is an M3x16mm and is adjustable in and out with a nut to lock it in place and depress the lever, this could be SHCS as well, I just like the idea of a rounded head to interface. The screw mounting the two pieces is an M3x8mm FHCS and it's in a slot and adjustable up and down. The two pieces have grooves to index together to prevent movement and are adjustable in 2.4mm increments, so the farthest you could align from perfect would be 1.2mm, this seemed a reasonable tradeoff between keeping standard 0.4mm nozzle settings and requiring a finer nozzle to print the parts. For 3mm backers this uses M3x10mm SHCSs, for thicker backers like MGN9 rails, you'll have to compensate for the thickness with a longer screw.
[image]
[image]

I will try to print these tonight when I get home to see how they do in both printability and interfacing with each other and my backer, but will not be able to test with the actual filament cutter for several weeks. I will confirm they at least interface together and with the backer and then upload part files if they do and you're welcome to test them after that.

erazor666, 2023-08-16:
Awesome, that is much cooler than any design I would have come up with. Absolutely a PR worthy. :)

juliusjj25, 2023-08-17:
Here it is set in place, not actually mounted but just set where it would go. I adjusted the height to the original 40mm of the 2020 extrusion so this could work with or without backers, MGN9 backers could work as long as the hole spacing is 10/20/40mm and the first hole is 5-15mm from the back edge. 
[image]
[image]
[depressor.zip](https://github.com/sorted01/Filametrix/files/12370989/depressor.zip)

juliusjj25, 2023-08-17:
If you want to print these, you'll need to reorient them to this position. The standard orientation of the axes between SS and Solidworks are different.... I need to figure out which way to model in Solidworks to get them to orient correctly in SS. I'll fix it for a final model.
[image]

juliusjj25, 2023-08-17:
I created a pull request, I'll admit, I have no clue if I did it right though!

shaneshort, 2023-08-26:
looks like this has been merged, should be ok to close the issue?

sorted01, 2023-08-26:
I merged it. Sorry was out for some time due to my baby 👶

closed 2023-09-07 by sorted01