sorted01/Filametrix #12: Will an 18mm spring work?

juliusjj25, 2023-08-14:
https://www.amazon.com/dp/B0BHHZBPW9/
https://www.amazon.com/dp/B089JYV7BT/

jfryman, 2023-09-07:
@juliusjj25 I tried out the [latter link](https://www.amazon.com/dp/B089JYV7BT/) last night, and had no luck with it as-is. The spring was just too long and fully compressed the knife wouldn't fully cut. I ended up trying to cut _only_ one side of the compressed end to some success.

The [first link](https://www.amazon.com/dp/B0BHHZBPW9/) won't arrive for me until Sunday, but I expect similar behaviors.

I'm also looking for a spring that might work better I can source from somewhere other than a pen spring (because that also seems unclear to me what sort of pen I should try and source this from).

juliusjj25, 2023-09-08:
I ended up ordering the second link and it is working just fine for me, however the hole for the spring is so tight it takes some work to get in in correctly.

jfryman, 2023-09-09:
Agreed. I had a similar issue and ended up grinding out just the slightest amount space to get the spring to seat. Very snug. It still wasn't as good as I hoped, and on a random shot I just bought this:

https://www.amazon.com/dp/B08169QK5X?psc=1&ref=ppx_yo2ov_dt_b_product_details

The spring for this is longer than requested (around 26mm) but it has the right amount of tension to have it spring right back in the way I'd expect.

juliusjj25, 2023-09-09:
It would be good at some point to have this design modified around a more commonly available spring geometry since this seems to be a pain point finding the right spring.

sorted01, 2023-09-09:
I have some springs lying around. Just need to find some time. Sorry 😣 will keep you posted

EmJay276, 2023-12-14:
> It would be good at some point to have this design modified around a more commonly available spring geometry since this seems to be a pain point finding the right spring.

I enlarged the hole for the spring in my design to 5mm and use a 5x0.4x30 [Spring](https://www.aliexpress.com/item/1005002652972568.html), I also use for my bowden tubes to prevent kinking.

Works perfectly fine for me, it's just a little bit difficult to get it in place at first.
[image]