sorted01/Filametrix #17: Build Volume Collision

juliusjj25, 2023-09-08:
How do you prevent the toolhead from colliding with the gantry mounted cutting point? It appears there is about a 50mm(Y) by 25mm(X) square in the back left corner where the toolhead could collide with the depressor.

gennro, 2023-09-09:
I would probably put your corner in the front left of your printer so it doesn't collide during homing operations and make sure none of of bed leveling etc go into that corner. Orcaslicer supports bed exclude zones as well along with nozzle volume purge compensation for handling the cut off piece of filament in the nozzle during a swap.

appleimperio, 2023-09-11:
@juliusjj25 I'm sorry to hack this post. Can you make a CW2 main buddy EBB SB2209 but without any sensor? just the ECAS. Thanks

juliusjj25, 2023-09-11:
> I would probably put your corner in the front left of your printer so it doesn't collide during homing operations and make sure none of of bed leveling etc go into that corner. Orcaslicer supports [...]

I figured it out, I created an STL of the bed with the corner notched out and uploaded that as my custom bed in SuperSlicer, now it won't let me place anything in that spot.
[image]
[image]

Now I'll just need to change QGL and bed mesh to remove that corner as well.

juliusjj25, 2023-09-11:
Here is the STL for a 300mm bed, I had to zip it to be able to post it here.
[Notched Bed.zip](https://github.com/sorted01/Filametrix/files/12580070/Notched.Bed.zip)

LostFool, 2023-11-22:
> Here is the STL for a 300mm bed, I had to zip it to be able to post it here. [Notched Bed.zip](https://github.com/sorted01/Filametrix/files/12580070/Notched.Bed.zip)

How did you get it to accept an origin of 0,0 at bottom left corner? I am trying to do a 350mm variant and Prusaslicer only accepts a custom one with origin at center of bed.

juliusjj25, 2023-11-22:
I use Superslicer and it worked just fine 0,0 should be the bottom left corner?

LostFool, 2023-11-22:
Prusaslicer is just dumb then. 

https://help.prusa3d.com/article/custom-bed-models_158431

I tried making the origin bottom left corner and I just get errors in prusaslicer.
Unfortunately I am stuck in Prusaslicer for the time being because I have a custom version that is more in-tune to purge temperatures of varying temperature filaments (PLA & ASA in same model)