sorted01/Filametrix #24: When using SB_CS2_Main_Buddy_EBB_ECAS_D2F big gap with default cable cover?

MaikelKelder, 2023-10-22:
Hi,

Not really an issue, but when using this SB_CS2_Main_Buddy_EBB_ECAS_D2F you leave a big gap open when using the default cable cover for the Stealthburner.
I wasnt able to find a adjusted one, I almost assume it exists?

Thanks for the MOD, its a beast! saving so much issues swapping filaments.

k5beason, 2023-11-07:
Same issue. Did you ever find the correct cover?

k5beason, 2023-11-08:
I believe the BTT files will work https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/Custom%20Printed%20Parts/cable_door_EBB_SB.stl

MaikelKelder, 2023-11-08:
I will check it out, thx!

k5beason, 2023-11-08:
> I will check it out, thx!
> 
> 
> 
> 
> 
> 
> > I believe the BTT files will work
> 
> > https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/Custom%20Printed%20Parts/cable_door_EBB_SB.stl

Let me know. On mine the bottom securing screw hole is about 1-2mm off still. I can't find any other covers that are close.

gabber130779, 2024-04-19:
same problem, but solved. I modified the cable door from BTT. you can fine it here.

https://www.printables.com/de/model/849383-cable-door-forstealthburner-cw2-filametrix-with-ec

MaikelKelder, 2024-04-19:
> same problem, but solved. I modified the cable door from BTT. you can find

Thank you!!

gabber130779, 2024-04-24:
> Thank you!!
> 
> 
> > same problem, but solved. I modified the cable door from BTT. you can find
> > it here.
> >
> >
> > https://www.printables.com/de/model/849383-cable-door-forstealthburner-cw2-filametrix-with-ec

Did it fit now?

MaikelKelder, 2024-04-25:
No, doenst fit, I guess the version with a double sensor in there is thicker or so.
Im using the OLD version of the CW2 motor plates, because of that, double sensors.
It also has a cutout for the 2209 canbus on the bottom.
I guess I have to get creative to get one fitting ;-)
[image]
[image]

MaikelKelder, 2024-04-25:
I am going to try this body, also double sensor, but no cutout and sensor is screwed on from bottom, much better.
Lets see if I can find a door for that, otherwise, ill make one and post it.

MaikelKelder, 2024-04-25:
Ok, I jus tprinted and fitted these and they work together. Your cover also fits but the hole for the screw is only just out of place, 2mm or so.

Vented Cover:
https://www.printables.com/model/772880-ebb-sb22402209-vented-cable-door-jan-2024-update/files

double sensor CW2 plate:
https://www.printables.com/model/826480-filametrix-body-cw2-for-sb2209sb2240-with-dual-fil
[image]