sorted01/Filametrix #25: HGX large gear cw2 remix

Darkmadda, 2023-11-08:
I finished my filametrix remix. I picked up a large gear (HGX) and i remixed a mainbody created by someone else that adds a ecas/filament-sensor, now we have a model that works with filametrix. https://www.printables.com/model/641383-clockwork2-large-gear-hgx-ecas-ercf-filament-senso . The model is tested by myself but i'd welcome anyone else to print and test it. I really like the filament sensor in this model as it uses the actual gears to sense the filament so you know exactly when the filament is in a gripped position.

bartoszwas, 2023-11-22:
If You would like to make version for LGX Lite I'm eager to test.

Darkmadda, 2023-11-23:
There is a LGX version on the printables page. it is untested so i'd love to know if it works

bartoszwas, 2023-11-23:
If You mean this one: https://www.printables.com/pl/model/576122-lgx-lite-stealthburner-filament-cutter there's no filament sensor and lots of supports required to print it. No idea is it compatible with EBB SB22xx.

Darkmadda, 2023-11-23:
No use the links in my first comment.
https://www.printables.com/model/641383-clockwork2-large-gear-hgx-ecas-ercf-filament-senso
No sports required. Filament sensor built in look and the model it is remixed from to see how the sensor works. The sensor works a little different than other sensors but it does work and works very well

bartoszwas, 2023-11-23:
Ah, Ok. I didn’t notice that there’s also a LGX version. I’ll check that for sure.

Darkmadda, 2023-11-23:
as for the sb22xx board. i have a sb2209 but there isn't much room for it with the large gear installed so it is a really tight fit. But it does fit (may require a little finagling )

Darkmadda, 2023-11-23:
I just realized you want to use the whole lgx extruder. my design just uses the gear set from that. you won't be able to use the whole extruder just the large gears from it. if that is acceptable then go for it but if you are looking to use the prebuilt extruder than it wont. To use the the whole unit it would require me to remix the design you first linked to.

bartoszwas, 2023-11-23:
Yes I was looking for solution to use the whole extruder, but, hey, if I can use gears only and it’ll work then why not.

Darkmadda, 2023-11-23:
Please let me know how it works. The lgx version is untested

bartoszwas, 2023-11-23:
I will. Just need to get some time to print and assemble - at least on Sunday I'll be back with feedback.

bartoszwas, 2023-11-24:
Got some info.

LGX gear seems to be a lil bit off of the filament path.
[image]

Rest parts are printing, but for now no other issues. Will be back with info when I assembly it.

Darkmadda, 2023-11-27:
A few notes:
1. "LGX gear seems to be a lil bit off of the filament path. Picture doesn't show it good... IMHO it should be like 0.2 mm deeper." 
- i'll look into adjusting that
2."I don't know how to install guilder. There's no hole for screw head... And IMHO a heat set insert would be nice there too" 
-there shouldn't be a screw there it should be there should be a steel shaft that press fits in (it should have been in the gear set).
[image]
3. "Next is plastic sprocket axe - in main body the hole is much bigger then that steel pin?" 
-there should be a bearing in there (again should have been in with the gears) is it possible for you to take a picture of all of the parts that came with your extruder (I know yours is not a gearset and is instead the kit so i don't know what you have) here is what my kit looks like 
[image]
5. "Filament can stuck on main body exit" 
-you are supposed to put a headset insert there (see breakaway here
[image]
6."Press fit parts doesn't fit so well."
 -I'll have to take mine apart and look at first picture. I know the part doesn't insert into the motor plate very far as for the second picture take the slot may be different from the latest CW2 release. they changed the shape of the motor plate slot. here in this picture 
[image]
(i know it's not of the large gear mod but it's extra parts i had laying around you can see that the slotted piece shape is much thinner on the left latest design. My remix is for the slightly older design so you will need the matching motor plate for that larger design) (i don't have a picture of the motor plate to show the actual slot). Make sure you are printing the correct motor plate. I'm guessing that is your problem. I've added the motor plates to the printables page so you can find them easier.

Darkmadda, 2023-11-27:
I just looked at the source design and the bearing i was talking about is a MR63ZZ bearing

bartoszwas, 2023-11-28:
So these are all parts I got from my lgx, three gears and three shafts. 

[image]

According to filament exit If you make cross section view you will see that it’s before the jest set insert. There’s a relief inside of the model…

Could you please share step file of main body with sensor? On printables there’s only without sensor and scad I cannot open in fusion 360. I wanna make some modification and test but for couple of days I won’t have access to my pc