sorted01/Filametrix #27: Supporting Rapido UHF / Rapido 2?

dflemstr, 2023-11-17:
I'm wondering if anyone has adapted to the new Rapido 2 series, and/or the UHF version with extended air channels?  Otherwise I might hack something together

bartoszwas, 2023-11-21:
Mine Rapido 2 fits in the old Rapido 1 stl. I’m not sure does every Rapido 2 will fit but mine does…

dflemstr, 2023-11-21:
Makes sense; however in any case the UHF version is very different and would require a special part for sure.

d0m1n1qu3, 2023-11-26:
but there are two versions in the voron git one for rapido 1 and one for rapido 2 .. there must be a difference

dflemstr, 2023-12-12:
Cool, in turn I hacked together a Rapido 2 UHF version with longer cooling ducts (the back part is the same, only the front part changed)
[fillametrix-rapido-2-uhf.zip](https://github.com/sorted01/Filametrix/files/13651783/fillametrix-rapido-2-uhf.zip)
This is a Blender hack of the model you posted above, because I couldn't find a manifold version of the original parts anywhere that would let me do the mod in OpenSCAD

BrewNinja, 2023-12-31:
Did you test your version @dflemstr  ? Im looking to do the same thing.

dflemstr, 2023-12-31:
@BrewNinja yes, I am currently using my version in a printer!

d0m1n1qu3, 2023-12-31:
and here my scad with fixed stls .. is also doing his job obn my brand new voron v2.4 :-D  but only HF not UHF
[SBRapidoV2FilametrixHF.zip](https://github.com/sorted01/Filametrix/files/13801229/SBRapidoV2FilametrixHF.zip)

gordon-quad, 2024-09-14:
Can we also have Rapido v1 UHF version please?

dflemstr, 2024-12-08:
I don't think we should share the hacked versions in the repo. It is better to use the original CAD to create a properly designed version in my opinion!

JarTheHead, 2024-12-08:
ah mb i was under the assumption the second one was done with CAD

dflemstr, 2024-12-09:
Well, technically, SCAD should count as CAD, of course, but I was hoping for a `.step` based version (or Fusion 360/whatever was used to produce the `.step` files) to retain the parametric curves, etc., from the original model.