sorted01/Filametrix #28: Support for Vz-HextrudORT Low

kriffo88, 2023-11-18:
Hello, will it be possible to have the mod to install a Vz-HextrudORT Low?? Thank you!

sorted01, 2023-11-19:
Yes. Currently testing it as I switched to hextrudort myself.

kriffo88, 2023-11-19:
Fantastic!! Can't wait to try it, thanks again! 👍🏼

kriffo88, 2023-12-06:
> Yes. Currently testing it as I switched to hextrudort myself.

Hi, is there any news?? Thank you!😃

sorted01, 2024-01-05:
Sorry took somewhat longer. Printing test parts right now

kriffo88, 2024-01-31:
> Sorry took somewhat longer. Printing test parts right now

Hi, is there any news on the release of the file? Thank you!

siparker, 2024-02-04:
+1 for Hextudort low version also.

kriffo88, 2024-02-29:
> Sorry took somewhat longer. Printing test parts right now

Hello! No news yet? Thank you

masterviana, 2024-06-07:
Hi, is there any updated on this topic? If this can fit on Vz-HextrudORT can also FIT on sherpa mini?  Can you share your work on this so we can go forward with de design?

kriffo88, 2024-06-19:
> Hi, is there any updated on this topic? If this can fit on Vz-HextrudORT can also FIT on sherpa mini? Can you share your work on this so we can go forward with de design?

I really think there will be no updates now...