sorted01/Filametrix #33: Issue with lost steps because of too much pressure while pressing the lever

seacred, 2023-12-04:
While cutting sometimes the AB Motors skip because of the high pressure applied on the lever.
Using nema 17 with 1A currently.
Did somebody noticed the same behaviour?

appleimperio, 2024-02-08:
do you manage to solve this? I'm having the same issue

seacred, 2024-02-09:
I wasn't able to solve this issue by now. i tried to reduce the cuttingmovespeed to a minimum but ist working for some cuts and suddenly it will skip. not reliable cutting over some swaps and longer prints possible by now.

slightlysketch, 2024-05-19:
so i had this issue too,
fixed it by doing a few different things

one i swapped out my 36-40 motors for some 42-40s 
next i changed the run current to 1.0

if youre still skipping motors after that, its a set up issue and your plunger might be too long.
need to be very precise

seacred, 2024-06-06:
fixed it by using a setting up a gcode, which turns up the motorcurrent during cutting and resets it to default afterwards

szsong, 2024-11-25:
@seacred could you please share your gcode please?

igiannakas, 2024-12-14:
You need something like this
`[gcode_macro _SET_CUT_CURRENT]
gcode:
    M400
    SAVE_GCODE_STATE NAME=SET_CUT_CURRENT_STATE_2
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT=1.9 HOLDCURRENT=1.9
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT=1.9 HOLDCURRENT=1.9
    M400
    G4 P1500
    M400
    RESTORE_GCODE_STATE NAME=SET_CUT_CURRENT_STATE_2
    M400`

igiannakas, 2024-12-14:
And a similar macro for setting the print current. Set these up and hook them up to the pre unload and post load happy hare user macro extension placeholders.