sorted01/Filametrix #35: TAP with Magnets

ThaNeub, 2023-12-12:
Is there maybe a possibility to add the Magnet holders for Voron TAP to the toolhead mounts? Maybe the screw for fixing the cutter arm could be moved to the front, so there would be room for the magnet mounts on the back?

EmJay276, 2023-12-14:
The toolheads do not contain the Tap magnet mounts, they are located at the carriage.

ThaNeub, 2023-12-14:
The CNC Version has the magnet holders integrated in the toolhead backside: https://github.com/Chaoticlab/CNC-Tap-for-Voron/blob/master/Community%20Mods/CNC-Tap-Magnets/hot%20ends/dragon.stl

EmJay276, 2023-12-14:
I have this [CNC Tap](https://github.com/Vitalii3D-xyz/MetalTap) / [Aliexpress clone](https://www.aliexpress.com/item/1005005708018536.html) where the magnets are mounted to the CNC part, like for the original version.

My opinion: Adding the magnets to the toolhead rear breaks the basic idea of modularity for the Stealthburner.

ThaNeub, 2023-12-14:
Okay it was a Suggestion. If not possible or it breaks the idea you could close it.

EmJay276, 2023-12-14:
I am not a maintainer, @sorted01 is :)