sorted01/Filametrix #42: Can't Find "Cutting Point on Gantry with Servo" Feature

Yanichar, 2023-12-24:
I was going through the Readme of this cool project (Project Name) and stumbled upon a mention of a "Cutting Point on Gantry with Servo." I tried to find this feature within the project's files, but no luck.

Could someone point me in the right direction or give a hint about where I might find more info about this?

murray484, 2024-01-14:
I also would like to find said file...

Thomsen1982, 2024-02-24:
Yes I am too.
Are there new information about it.

Marucci87, 2024-07-21:
I found this on thingyverse: https://www.thingiverse.com/thing:6584232 I have just printed it and will try it out within the next few days, looks like you just have to uncomment the first few lines and a few in the cutting macros in the additional.cfg. I just hope the stls I linked above will reach to my cutter....Im using the g2 extruder version. I assume if it doesnt line up I could just scale the stl for the servo mount taller or shorter.

Thomsen1982, 2024-07-21:
It looks interesting, it’s on the list for the next print 😂