sorted01/Filametrix #44: Mounting instruction for V6-R6 Hotend

TheMaxx2804, 2024-01-10:
Hi there,

i am missing the mounting / building instruction for the V6_R6 Hotend.
As shown in the picture the hotend is standing out some mm from the print head back and front piece.
In this case i do not know how the knife would be mounted to work correctly.

Thank you in advance for any suggestions.

[image]

mole6999, 2024-04-04:
I could use help woth the same problem. Has anyone figures it out yet?

protorayn, 2024-05-16:
Yes. remove the black plastic ptfe  adapter. I removed it and printed a sleeve that holds the ptfe in place for cutting.

ThunderClap86, 2024-12-19:
> > printed a sleeve that holds the ptfe in place for cutting.
> 
> So the cutting knife has to be cut in length that it does stay on the silver collar and that it could not flip down that collar.
Good suggestions on both of these. 

Have a STL or CAD item for the collar you made to replace the pressfit collar in the pic above?

ThunderClap86, 2024-12-19:
> Yes. remove the black plastic ptfe adapter. I removed it and printed a sleeve that holds the ptfe in place for cutting.

Guess it's you I need to be asking for the file(s) if you are willing to share.

protorayn, 2024-12-19:
I'll put it up on printables when I get home tonight

protorayn, 2024-12-19:
> > Yes. remove the black plastic ptfe adapter. I removed it and printed a sleeve that holds the ptfe in place for cutting.
> 
> Guess it's you I need to be asking for the file(s) if you are willing to share.

https://www.printables.com/model/914674-filament-guide-sleeve-for-dragonfly-bmo

This is the sleeve I made for replacing the bowden collet.

alfadex, 2025-04-19:
@TheMaxx2804  Says the problem is here as you can see in the photo. Metal hotend is 1 mm above printed part and the cutter stop to the metal. Has somebody find any solution?

[image]