sorted01/Filametrix #45: Orbiter 2 mount not working

appleimperio, 2024-01-26:
After mounting the filametrix in my orbiter 2 with the compatible mount by emjay276 The level is not moving at all because the mount  interfere.  
[image]

I will try this model that looks to have the clearance. https://www.printables.com/model/647601-orbiter-filametrix-mod

SaltyPaws, 2024-02-06:
I have added a pull request with 2 commits that addresses this issue, and with PG7, standard and sensor  version.

gonchogsm, 2024-05-10:
I had the same problem. I will print SaltyPaws files now. Hope they are merged because current Orbiter file is useless for filametrix