sorted01/Filametrix #46: Question about Voron SW

rollduck, 2024-01-29:
Good day, 

I'm a Voron Switchwire owner and due to the design of this printer, I have to put a depressor unit on the z-axis instead of the y-axis and I don't know how to configure it. I've changed "variable_pin_loc_y" to "variable_pin_loc_z" but of course it doesn't work :) There is any possibility to use depressor mounted on z axis?

Thanks in advcance
Best, 

Radek

appleimperio, 2024-02-08:
You just ignore the "variable_pin_loc_y" I set it up to half of the bed size just make sure the variable_pin_loc_x is correct. I'm having other issue right now but I think is skipping steps when hitting the filament.

slightlysketch, 2024-05-19:
> You just ignore the "variable_pin_loc_y" I set it up to half of the bed size just make sure the variable_pin_loc_x is correct. I'm having other issue right now but I think is skipping steps when hitting the filament.

if youre skipping steps,
make sure youre running 42-40 motors on whatever axis does the motion to hit the plunger.
second set your run current to 1.0
last dont use stealthchop (this can possibly be ignored, but i had to turn if off)

appleimperio, 2024-05-19:
@slightlysketch Thanks for your suggestion. I'm using the EREC now and it's working great.