sorted01/Filametrix #56: CAD Files for EBB and D2F Options

eprouty, 2024-08-31:
I want to update the EBB and D2F options from the old motor plate, in order to support the new motor plate and M6 bowden couplers.  However, the CAD files for these old motor variants are missing.

Does anyone have these they can add? or know where I can find them?

eprouty, 2024-08-31:
@EmJay276 Do you happen to have these? Saw you had an open PR for other new motor plate STLs

EmJay276, 2024-09-03:
Did you check my changes?
https://github.com/EmJay276/ERCF_v2/blob/CW2_new_motor_plate/Recommended_Options/ERF_Filament_Cutter/README.md
I already modified the versions for the `1_Toolhead_And_Entry_Sensors`, `2_Toolhead_Sensor (EBB Board)` and `3_Entry_Sensor`

I think they were only available as STL when I modified them. So I could just do a boolean operation to remove the additional material. I attached the parts I removed from the STLs, if you need to modify other parts.
One has an Offset to fix some issues concerning the triangulation of the STL where tiny parts were left after the boolean operation.
[CW2_cutoffs_old_motor_plate.zip](https://github.com/user-attachments/files/16858855/CW2_cutoffs_old_motor_plate.zip)